openenergymonitor/EmonScripts #6: Rotated logs review

TrystanLea, 2019-08-07:
Screenshots of my rotated logs based on the logrotate and log2ram configuration on the present release:

[image]

[image]

[image]

Is this number of rotated logs based on the 100k rotation unwieldy? too many files to open and work through, if you wanted to find an error?

pb66, 2019-08-11:
Both the emonhub and emoncms logrotate config drop-ins should have their own `maxsize=` settings.

Even 1M was always way too small for emonhub, 3-5M maybe?

and the apache logs should also be getting sent to emoncms log folder by the vhost and then being rotated at the higher maxsize into /var/log.old/emoncms. Why are they in /var/log.old?

TrystanLea, 2019-08-23:
Hmm I'm not sure why the apache logs are there, the vhost settings direct them to /var/log/emoncms and there are also there - though not gzip'ed..

quite a few files over 100k too..

[image]

TrystanLea, 2019-08-23:
File permissions where wrong on 00_defaults!

TrystanLea, 2019-08-23:
The apache2 log not in the emoncms folder is created by this entry in conf-enabled:

[image]

TrystanLea, 2019-08-23:
I've changed it to /var/log/emoncms/apache2-error.log there for now and fixed the permissions on 00_defaults

TrystanLea, 2019-08-25:
I've changed the maxsize on emoncms and emonhub logs to 3M

pb66, 2019-08-25:
>I've changed it to /var/log/emoncms/apache2-error.log there for now

That sounds like it's the right place to set the default error log on an emonSD permanently.

>fixed the permissions on 00_defaults

What is "00_defaults" ?