openenergymonitor/EmonScripts #31: Running pip as sudo

borpin, 2019-09-13:
https://github.com/openenergymonitor/EmonScripts/blob/e22ed870790246c5733817d3cb1b46e5ccd364b5/install/redis.sh#L20

I think `pip` should **not** be run with `sudo`.

borpin, 2019-09-15:
On testing on Ubuntu, the service (such as `service-runner`) that uses the python-redis package requires the `user` setting to match the user that installed it.

Installing the package as `sudo` means that the package is usable by any user.

This is not strictly good practice, but changing it may break too much.

TrystanLea, 2019-09-16:
I will make a note to test on next build without sudo

borpin, 2019-09-16:
It would mean that the `service-runner` User would need to be set to the user who installed emoncms. I think it will break too much (it should work on the pi).

TrystanLea, 2019-09-16:
ah I see what you mean, the idea of a dedicated user as you suggest is an interesting one

borpin, 2019-09-16:
I think this was floated before probably by Paul.