openenergymonitor/EmonScripts #32: Folder is used in config but does not exist

borpin, 2019-09-15:
https://github.com/openenergymonitor/EmonScripts/blob/a57d5b8330719bc7802cccc174d75d337b75956a/install/emoncms_modules.sh#L67

This location is used as a config item but it does not exist.

`/opt/openenergymonitor/data/uploads`

Folder needs to be checked for existence and if not, created and configured for use (chown www-data:USER).

TrystanLea, 2019-09-16:
It should be created by the backup install script here: https://github.com/emoncms/backup/blob/master/install.sh

but looks like it created it in /opt/emoncms/data/uploads on my test system here..

TrystanLea, 2019-09-16:
This commit should fix it https://github.com/openenergymonitor/EmonScripts/commit/6b26c199ae1b2c78c4ba1ce00dcfcd9b4e0a8227

borpin, 2019-09-23:
An Import and then a Backup does now work.

However, it seems now that the backup and import functions use the same folder so there is no distinction between the 2 (so an import followed by an export) will overwrite the 'import' file.

Suggest that a time is added to the filenames to make them unique.

Below is the `backup/config.cfg` and directory listings - they do not seem to be consistent.

```
brian@ubunt-emon:/opt/openenergymonitor$ cat /opt/emoncms/modules/backup/config.cfg
# Emoncms Export and Import scripts config file

## DEFAULT CONFIG ##

# These settings are used for both the export and import scripts.
# To export from one location and import into another
# Used by both PHP and Bash and comments are not ignored

# Change all capitalised placeholders to relevant locations

# user
user="brian"

backup_script_location="/opt/emoncms/modules/backup"

# Directory location of Emoncms
emoncms_location="/var/www/emoncms"

# Destination location of exported backup .tar.gz - Must be writable
backup_location="/opt/openenergymonitor/data"

# location of phpfina phpfiwa and phptimeseries databases
database_path="/var/opt/emoncms"

# Location of emonhub.conf - leave blank if backup not required
emonhub_config_path="/etc/emonhub"

# Location of specimen emonhub config
emonhub_specimen_config="/opt/openenergymonitor/emonhub/conf"

# Location of compressed backup data to import .tar.gz
backup_source_path="/opt/openenergymonitor/data/uploads"
```

```
brian@ubunt-emon:/opt/openenergymonitor$ ls -la /opt/openenergymonitor/
total 16
drwxr-xr-x 4 brian root  4096 Sep 23 06:50 .
drwxr-xr-x 4 root  root  4096 Sep 23 06:24 ..
drwxrwxr-x 4 brian brian 4096 Sep 23 07:42 data
drwxrwxr-x 9 brian brian 4096 Sep 23 06:25 EmonScripts
```

```
brian@ubunt-emon:/opt/openenergymonitor$ ls -la /opt/openenergymonitor/data/
total 19728
drwxrwxr-x 4 brian    brian     4096 Sep 23 07:42 .
drwxr-xr-x 4 brian    root      4096 Sep 23 06:50 ..
-rw-r--r-- 1 root     root  20152875 Sep 23 07:41 emoncms-backup-2019-09-23.tar.gz
-rw-r--r-- 1 root     root     26563 Sep 23 07:41 emoncms.sql
drwxr-xr-x 3 brian    root      4096 Sep 23 07:37 import
drwxrwxr-x 2 www-data brian     4096 Sep 23 07:37 uploads
```

```
brian@ubunt-emon:/opt/openenergymonitor$ ls -la /opt/openenergymonitor/data/import/
total 20
drwxr-xr-x 3 brian    root     4096 Sep 23 07:37 .
drwxrwxr-x 4 brian    brian    4096 Sep 23 07:42 ..
drwxr-xr-x 2 www-data root     4096 Sep 20 12:15 phpfiwa
-rwxrwxr-x 1 www-data www-data 6987 Jul 18 18:27 settings.php
```
```
brian@ubunt-emon:/opt/openenergymonitor$ ls -la /opt/openenergymonitor/data/uploads/
total 8
drwxrwxr-x 2 www-data brian 4096 Sep 23 07:37 .
drwxrwxr-x 4 brian    brian 4096 Sep 23 07:42 ..
```

closed 2019-10-02 by borpin