openenergymonitor/EmonScripts #34: Create scheme for default configs

borpin, 2019-09-16:
The current `config.ini` file needs to be excluded from the repo via `.gitignore` so changes do not prevent a later update of the repo.

I suggest that by default there is no `config.ini`. 

I suggest there is a `pi.config.ini` and perhaps a `linux.config.ini` with settings likely to work for a Linux system other than Raspbian.

On running `init.sh` if the question is no review, a pi is assumed and the `pi.config.ini` is copied to `config.ini` and install proceeds.

If a review is required, the user is told to copy either of the base ini files and modify as required.

As an aside, I also think some of the later True/False options need to be brought to the top and the modules bit pushed down.

TrystanLea, 2019-09-17:
Implemented this here:
https://github.com/openenergymonitor/EmonScripts/commit/08afb079427219364058b07797a73535e7c43622 

and
https://github.com/openenergymonitor/EmonScripts/commit/3349eda148d977651d433f624f3aeb28a293acb0

emonsd.config.ini is copied to config.ini at the start of main.sh if config.ini does not exist regardless. it then asks whether you would like to edit the config.ini in place

borpin, 2019-09-17:
OK, can you add `config.ini` to a `.gitignore` file please?

TrystanLea, 2019-09-17:
yes done in the first commit there :)

borpin, 2019-09-17:
Oops - sorry - missed it :man_facepalming:

closed 2019-09-17 by TrystanLea