openenergymonitor/EmonScripts #35: Non-Pi Logrotate

borpin, 2019-09-16:
A marker for this

https://community.openenergymonitor.org/t/emonsd-next-steps-filesystem-logrotate/10693/199

TrystanLea, 2019-09-17:
Implemented, thanks for the example config, see:

https://github.com/openenergymonitor/EmonScripts/commit/1eafefa4d84deeb90a66335c541da5234174585f

https://github.com/openenergymonitor/EmonScripts/commit/39398d6fb22025ec455c4a2f8bbdbfad3761a03c

borpin, 2019-09-17:
Is this right?

https://github.com/openenergymonitor/EmonScripts/blob/3349eda148d977651d433f624f3aeb28a293acb0/install/non_emonsd.sh#L2

Looks ideal.

TrystanLea, 2019-09-17:
Fixed thanks: https://github.com/openenergymonitor/EmonScripts/commit/b804e32b6c644eea42087f623ed0bed4cc922d0c

closed 2019-09-17 by TrystanLea