openenergymonitor/EmonScripts #36: EmonPiFileSystem explanation - links to services

borpin, 2019-09-17:
@alexandrecuer 

Firstly this is excellent!

I think this should be just 'EmonCMS File System' as it is not specific to the Pi (with specific Pi stuff noted later).

Secondly, I don't think this statement is accurate (couldn't find a better way of linking to it).
[all systemd services have to be symlinked in /etc/systemd/system](https://github.com/openenergymonitor/EmonScripts/blob/master/EmonPiFileSystem.md#all-systemd-services-have-to-be-symlinked-in-etcsystemdsystem)

What actually happens is that the install script creates a link, for the service file to `/lib/system/systemd/`
https://github.com/openenergymonitor/EmonScripts/blob/776c0e0d8f5e5c3d2a95873f26690e3e44725332/common/install_emoncms_service.sh#L53

The command `sudo systemctl enable <service` is the process that then creates the link in `/etc/systemd/system`. Thus, if the user wants to override this service, they can.

It's quite an important distinction :grinning:.

I've found a few things that will need specific differences to support non-pi environments.

alexandrecuer, 2019-09-17:
Yes I agree, the install script is creating the link...
I just wanted to explicit that if yu do not have the symlink, things will not work (english is not my mother tongue)
what do yu think of simply ; 
all systemd services are symlinked in /etc/systemd/system

borpin, 2019-09-17:
Rest assured your English is very good.

I'll propose a change. At the moment it suggests the symlinks are created by the script in that location which is misleading.