openenergymonitor/EmonScripts #42: Substitue emoncms_datadir in settings.ini file.

borpin, 2019-10-02:
Currently the datadir is hard coded into the `emonpi.settings.ini` so takes no account of a setting in the `install/config.ini`

Use `$emoncms_datadir` in the same way as the other directories are used.

https://github.com/openenergymonitor/EmonScripts/blob/b255689d378f9ab30447f183635413ec56d3d163/install/emoncms_core.sh#L34

This should be done here:
https://github.com/openenergymonitor/EmonScripts/blob/b255689d378f9ab30447f183635413ec56d3d163/install/emoncms_core.sh#L45

If only 2 folders are to be referenced in the `settings.ini` then consideration should be made as to whether they are created.

Could there be an engine selection process as part of the install?

TrystanLea, 2019-10-02:
As a starting point I have added the application of the config.ini datadir to the settings.ini file:

https://github.com/openenergymonitor/EmonScripts/commit/b8572cfa3804d851df088434b3023c39f814f1fa

Lets keep this one open for future improvement

TrystanLea, 2020-01-23:
Is the remaining item here?

> Could there be an engine selection process as part of the install?

borpin, 2020-01-24:
I thin drop that and close this.

closed 2020-01-24 by TrystanLea