openenergymonitor/EmonScripts #44: Installation on a DigitalOcean droplet

TrystanLea, 2019-10-02:
1\. Click on 'Create' to create a new cloud server
2\. Select Ubuntu, standard $5/month, region e.g London then Create Droplet
3\. DigitalOcean email root login details, ssh in using these credentials.
4\. Change root password
5\. Follow [Initial Server Setup with Ubuntu 18.04](https://www.digitalocean.com/community/tutorials/initial-server-setup-with-ubuntu-18-04)
6\. Disable root login: 

    sudo nano /etc/ssh/sshd_config
    PermitRootLogin no
    sudo service ssh restart

7\. Start the script

    wget https://raw.githubusercontent.com/openenergymonitor/EmonScripts/master/install/init.sh
    chmod +x init.sh
    ./init.sh

8\. Modify config.ini for non emonSD installation.

Disable mosquitto:

    install_mosquitto=false
    install_mosquitto_server=false
    install_mosquitto_client=false

Disable emonhub installation

    install_emonhub=false

Disable firmware, emonpilcd, emonsd, wifiap installation

    install_firmware=false
    install_emonpilcd=false
    install_emonsd=false
    install_wifiap=false

Set user as applicale and set emonSD_pi_env=0

    user=oem
    hostname=emonpi
    emonSD_pi_env=0

Comment out wifi and demandshaper module:

    #emoncms_modules[wifi]=stable
    #symlinked_emoncms_modules[demandshaper]=stable

9\. Run main installation script:

    ./main.sh

10\. .... still in progress ...

TrystanLea, 2019-10-02:
Issues:

- non_emonsd.sh did not run
- emoncms register does not disappear after first user creation
- services

TrystanLea, 2019-10-02:
2nd test, with issues above fixed now works a treat!

Remaining issue: emoncms update UI includes items not installed: firmware & emonhub

TrystanLea, 2019-10-02:
@borpin I've integrated the service file modifications using the drop-ins to /lib/systemd/.. all working well

borpin, 2019-10-02:
> * emoncms config module should not be installed

> 2nd test, with issues above fixed now works a treat!

Did you comment the modules out or is there a logic test?

My last install, I found WiFi installed unless I commented that module out too.

TrystanLea, 2019-10-02:
Yes I commented them out. I will work on improved logic in the next version

TrystanLea, 2019-10-03:
Latest install test on digitalocean went well, no issues everything running as expected.

borpin, 2019-10-03:
Perhaps a 'Digital Ocean' sample config file might help in the repo especially as it will not just be changing values?

borpin, 2019-10-07:
Added to documentation. #51

closed 2020-01-23 by TrystanLea