openenergymonitor/EmonScripts #46: Errors on update

borpin, 2019-10-02:
On a non-pi system, the following errors were seen on update of Emoncms

```
sudo: /opt/openenergymonitor/emonpi/lcd/emonPiLCD_detect.sh: command not found
sudo: /opt/openenergymonitor/emonpi/lcd/emonPiLCD_detect.sh: command not found
/opt/openenergymonitor/EmonScripts/update/main.sh: line 33: [: ==: unary operator expected
/opt/openenergymonitor/EmonScripts/update/main.sh: line 33: [: ==: unary operator expected
```
No rfm module is fitted (not a Pi).
```
Hardware detected: rfm2pi
```
```
Start emoncms update:
-------------------------------------------------------------
Update Emoncms Core
-------------------------------------------------------------

Traceback (most recent call last):
  File "/opt/openenergymonitor/EmonScripts/common/./md5sum.py", line 16, in 
    with open(file_name) as file_to_check:
IOError: [Errno 2] No such file or directory: '/var/www/emoncms/settings.php'
current settings.php md5: 
Default settings.php md5: 59e80f731935867f5f93753d49c0fe86
settings.php HAS been user modified
```

How does EmonCMS know what hardware it is on and what subsystems have been installed?

TrystanLea, 2019-10-03:
Yes I need to review that, noticed the same

borpin, 2019-10-03:
This PR will fix the LCD bit (and gets rid of the unary operator bit as well).

https://github.com/emoncms/emoncms/pull/1440

closed 2019-10-16 by borpin