openenergymonitor/EmonScripts #49: non-pi logrotate file permissions incorrect

borpin, 2019-10-04:
https://github.com/openenergymonitor/EmonScripts/blob/b804e32b6c644eea42087f623ed0bed4cc922d0c/install/non_emonsd.sh#L8

On running the logrotate on Ubuntu the following error is seen
```
brian@ubunt-emon:/var/log/emoncms$ sudo /usr/sbin/logrotate /etc/logrotate.conf
error: Ignoring emoncms because of bad file mode - must be 0644 or 0444.
```

But I could not fix it for some reason...

TrystanLea, 2019-10-04:
seems to be working here?

[image]

borpin, 2019-10-05:
Ran a pi install yesterday.

```
 $ ls -la /opt/openenergymonitor/EmonScripts/defaults/etc/logrotate.d/
total 24
drwxr-xr-x 2 pi   pi 4096 Oct  4 21:52 .
drwxr-xr-x 5 pi   pi 4096 Oct  4 21:52 ..
-rw-r--r-- 1 root pi   13 Oct  4 21:52 00_defaults
-rw-r--r-- 1 root pi  118 Oct  4 21:52 emoncms
-rw-r--r-- 1 pi   pi  190 Oct  4 21:52 emoncms-non-emonsd
-rw-r--r-- 1 root pi  184 Oct  4 21:52 emonhub
```
on Ubuntu (note I have manually changed the `emoncms-non-emonsd` file).
```
$  ls -la /opt/openenergymonitor/EmonScripts/defaults/etc/logrotate.d/
total 24
drwxrwxr-x 2 brian brian 4096 Oct  4 14:41 .
drwxrwxr-x 5 brian brian 4096 Oct  4 14:41 ..
-rw-rw-r-- 1 brian brian   13 Oct  4 14:41 00_defaults
-rw-rw-r-- 1 brian brian  118 Oct  4 14:41 emoncms
-rw-r--r-- 1 root  brian  190 Oct  4 14:41 emoncms-non-emonsd
-rw-rw-r-- 1 brian brian  184 Oct  4 14:41 emonhub
```

Must be a Raspbian thing.  Suggest a `chmod` is added to the folder in the `non-emonsd.sh` script.

closed 2019-10-22 by borpin