openenergymonitor/EmonScripts #50: Write load from php sessions

TrystanLea, 2019-10-05:
Not sure why I didnt notice this in previous testing but there seems to be quite a bit of write load resulting from updating php sessions. Placing the session folder in a tmpfs makes a big difference:

    tmpfs           /var/lib/php/sessions  tmpfs   nodev,nosuid,size=1M,mode=1777  0  0

TrystanLea, 2019-10-05:
added to master: https://github.com/openenergymonitor/EmonScripts/commit/79892bd172d783150d0590408b22c1ed4b7a9b56

TrystanLea, 2019-10-05:
should we symlink fstab or copy as part of update process? any thoughts @borpin

borpin, 2019-10-05:
Not sure if a symlinked fstab works - not my area really.

I'd append the line to the fstab if it is a pi/tmpfs system.

Is this a new problem? What has introduced it? The vhost?

closed 2020-01-23 by TrystanLea