openenergymonitor/EmonScripts #67: Review partition sizes

borpin, 2019-10-28:
re PR #66 (documentation)
re Issue #58 

https://community.openenergymonitor.org/t/emonsd-17oct19-image-partition-sizes/12282/

The current script/image fixes the size of the `rootfs` and only works for 16GB cards.

Suggest the script is amended to fix the Ext2 partition to the end of the card at a max size.

```shell
#EmonSD Fix end of Rootfs
if [ $ROOT_DEV_SIZE -gt 16000000 ]; then
  #16GB card or above, assign last 10GB for data 
  TARGET_END=$(($ROOT_DEV_SIZE - 20972544 - 1 ))
elif [ $ROOT_DEV_SIZE -gt 8000000 ]; then
  #8GB card, assign last 4GB for data 
  TARGET_END=$(($ROOT_DEV_SIZE - 7341056 - 1 ))
fi
```

More discussion on community thread.

I'm not certain this is a robust solution.

TrystanLea, 2019-10-30:
The old scripts for emonSDexpand are located here https://github.com/emoncms/usefulscripts/tree/master/sdpart not sure if these provide some useful pointers? I will ask Glyn about these.

borpin, 2019-10-30:
Fist thing - are the sizes appropriate for the data partition, I think they probably are.

I did find the script eventually!

borpin, 2019-10-31:
> Is it possible to just assign a fixed 4GB for the root partition and then calculate whatever space there is left for the data partition?

It is, but I don't think that is a good use of the SD card. Will anyone ever get more than 10GB data?

Limiting the size of rootfs could prevent a user using the Pi for other things.

What does (from memory) `sudo parted print` give?

Is this a burned image or a image created from scratch? There may be a difference when the 'image' file is created.

TrystanLea, 2019-11-01:
Fair point re limiting size. sudo parted print does not give me any info?
Im using a burned image

I just shrunk a 16gb image to a data partition size of 4gb, and now im expanding using the sdpart_imagefile script

Here's the print out so far:

    pi@emonpi:/opt/emoncms/modules/usefulscripts/sdpart $ sudo ./sdpart_imagefile 
    ./sdpart_imagefile: line 29: rpi-rw: command not found

    ======================================================

    Current Disk Info:
    Disk /dev/mmcblk0: 14.4 GiB, 15476981760 bytes, 30228480 sectors
    Units: sectors of 1 * 512 = 512 bytes
    Sector size (logical/physical): 512 bytes / 512 bytes
    I/O size (minimum/optimal): 512 bytes / 512 bytes
    Disklabel type: dos
    Disk identifier: 0x6c586e13

    Device         Boot   Start      End Sectors  Size Id Type
    /dev/mmcblk0p1         8192   532479  524288  256M  c W95 FAT32 (LBA)
    /dev/mmcblk0p2       532480  8959999 8427520    4G 83 Linux
    /dev/mmcblk0p3      8960000 17348607 8388608    4G 83 Linux

    ======================================================

    Proposed changes to be made:
     SD card total disk size = 14.4140625Gb
     Data Partition size     = 10.0927739143Gb

    Are you sure you want to proceed? [Y/n] y
    ./sdpart_imagefile: line 61: rpi-rw: command not found
    Created symlink /etc/systemd/system/resize2fs_once.service → /lib/systemd/system/resize2fs_once.service.
    Created symlink /etc/systemd/system/multi-user.target.wants/resize2fs_once.service → /lib/systemd/system/resize2fs_once.service.

    Welcome to fdisk (util-linux 2.33.1).
    Changes will remain in memory only, until you decide to write them.
    Be careful before using the write command.

    Command (m for help): Partition number (1-3, default 3): 
    Partition 3 has been deleted.

    Command (m for help): Partition type
       p   primary (2 primary, 0 extended, 2 free)
       e   extended (container for logical partitions)
    Select (default p): Partition number (3,4, default 3): First sector (2048-30228479, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (8960000-30228479, default 30228479): 
    Created a new partition 3 of type 'Linux' and of size 10.1 GiB.
    Partition #3 contains a ext2 signature.

    Command (m for help): 
    Disk /dev/mmcblk0: 14.4 GiB, 15476981760 bytes, 30228480 sectors
    Units: sectors of 1 * 512 = 512 bytes
    Sector size (logical/physical): 512 bytes / 512 bytes
    I/O size (minimum/optimal): 512 bytes / 512 bytes
    Disklabel type: dos
    Disk identifier: 0x6c586e13

    Device         Boot   Start      End  Sectors  Size Id Type
    /dev/mmcblk0p1         8192   532479   524288  256M  c W95 FAT32 (LBA)
    /dev/mmcblk0p2       532480  8959999  8427520    4G 83 Linux
    /dev/mmcblk0p3      8960000 30126080 21166081 10.1G 83 Linux

    Command (m for help): The partition table has been altered.
    Syncing disks.

    This error message can however be disregarded, because your system
    is about to be rebooted, and the new partition table will then be
    read by your operating system.

    Writing the resize2fs_once script

    ======================================================================

    So far, so good... in 20s your system will reboot
    and resize the /home/pi/data filesystem to fill the resized partition.
    **THIS OPERATION WILL TAKE UP TO 20 MINUTES**
    When completed the Pi will poweroff and close down.
    PLEASE DO NOT UNPLUG THE POWER FOR AT LEAST 30min
    because it will likely result in an unusable filesystem.

    ======================================================================
    Connection to 192.168.0.103 closed by remote host.

TrystanLea, 2019-11-02:
This worked fine:

    pi@emonpi:~ $ df -h
    Filesystem      Size  Used Avail Use% Mounted on
    /dev/root       4.0G  1.9G  2.0G  49% /
    devtmpfs        484M     0  484M   0% /dev
    tmpfs           488M     0  488M   0% /dev/shm
    tmpfs           488M  6.5M  482M   2% /run
    tmpfs           5.0M  4.0K  5.0M   1% /run/lock
    tmpfs           488M     0  488M   0% /sys/fs/cgroup
    tmpfs            30M     0   30M   0% /tmp
    tmpfs           1.0M     0  1.0M   0% /var/lib/php/sessions
    tmpfs           1.0M     0  1.0M   0% /var/tmp
    /dev/mmcblk0p3   10G  3.5M  9.5G   1% /var/opt/emoncms
    /dev/mmcblk0p1  253M   52M  201M  21% /boot
    log2ram          50M  1.7M   49M   4% /var/log
    tmpfs            98M     0   98M   0% /run/user/1000

TrystanLea, 2019-11-02:
So I think the steps now are to:

1. Remove the errors associated with rpi-rw
2. Edit "and resize the /home/pi/data filesystem to fill the resized partition." to "and resize the data filesystem to fill the resized partition."
3. Create a symlink to emonSDexpand in EmonScripts
4. Include in update mechanism for users downloading the image

Should we consider shipping the image with a smaller data partition to cater for 8GB sd cards?

borpin, 2019-11-02:
> 2\. Edit "and resize the /home/pi/data filesystem

Why?  Emoncms now uses the `/opt/emoncms` structure.

So does this move the Ext2 partition to the end?  If it does, I think the rest should just be rootfs.

> Should we consider shipping the image with a smaller data partition to cater for 8GB sd cards?

Can do. Just need to move *and* resize the Ext2 partition then.

TrystanLea, 2019-11-06:
I've made the above changes and created a new set of stable releases today for all emoncms modules with recent changes.

Testing the latest EmonScripts it got stuck to start with on init_resize.sh. I reverted your recent changes and it started working fine again.

The EmonScripts install script then completed a full installation without any apparent issues.

The emonSDexpand command is now present:

[image]

> So does this move the Ext2 partition to the end? If it does, I think the rest should just be rootfs.

No I think it just expands the data partition to the full extent. I see your point about increasing the root partition size and Im open to that in future versions.

TrystanLea, 2019-11-06:
The emonSDexpand is not yet available via update on the current image, I will look into adding that next.

borpin, 2019-11-06:
> Testing the latest EmonScripts it got stuck to start with on init_resize.sh.

What size card?

TBH I had not been able to test it as I'm out of spare cards especially 32Gb ones.

TrystanLea, 2019-11-06:
On our standard 16GB card

TrystanLea, 2019-11-06:
I tried it twice and both times after the reboot the pi failed to come back. When power cycled the pi manually it did come back but without a data partition /dev/mmcblk0p3

borpin, 2019-11-06:
I'll have a go and see what I have done wrong 😁

alexandrecuer, 2020-03-19:
dont know if I have to post here but I had pb with the new raspbian image ( 2020-02-13) using the init_resize script
/dev/mmcblk0p3 is not created anymore..i had to do it by hand using gparted on a separate linux machine...

borpin, 2020-03-20:
Thanks. I need to find time to look at this again.

borpin, 2020-05-01:
Wow over a year - still got there in the end

```bash
  # Dev size divisible by 2048
  ROOT_DEV_SIZE2048=$((($ROOT_DEV_SIZE / 2048) * 2048 ))

  #EmonSD Fix end of Rootfs
  if [ $ROOT_DEV_SIZE -gt 16000000 ]; then
    #16GB card or above, assign last 10GB for data
    TARGET_END=$(($ROOT_DEV_SIZE2048 - 20971520 - 1 ))
  elif [ $ROOT_DEV_SIZE -gt 8000000 ]; then
    #8GB card, assign last 4GB for data
    TARGET_END=$(($ROOT_DEV_SIZE2048 - 7340032 - 1 ))
  else
    # Assign 2GB to rootfs
    TARGET_END=4098047
  fi
```

I'll do a PR

borpin, 2023-02-10:
Close as method no longer is appropriate.

closed 2023-02-10 by borpin