openenergymonitor/EmonScripts #81: /var/log filling up

borpin, 2020-04-09:
`/var/log` is filling up again. https://community.openenergymonitor.org/t/var-log-filling-up-again/13808/

This is because the `logrotate.log` file is not being rotated!

I have added a rotation spec into the `00_defaults` file to see what happens. Not sure how it will react to rotating itself.

However, you need to 
```
sudo truncate -s 0 /var/log/logrotate/logrotate.log
```
the file first otherwise the logrotate fails as there is not enough room to write the file.

Will need to be added to the update script.

TrystanLea, 2020-04-09:
Thanks @borpin good to know, how long did this take to fill up, I dont think I've noticed this?

borpin, 2020-04-09:
It does take a while but we have had one report on the community I think.

In the last 5 hours the log has grown about 8k per hour I'd estimate

It will be a bit insidious as the log grow slowly. When the problem occurs depends on how much you are logging other things.

I'd suggest a `truncate` command is added to the update script to at least mitigate it.

borpin, 2023-01-29:
#159 opened

closed 2023-01-29 by borpin