openenergymonitor/EmonScripts #92: raspbian issues ?

alexandrecuer, 2020-06-18:
Just to share because I faced some kernel related issues when creating a new SD from scratch....would like to know if somebody else was facing similar problems

using the last raspbian image : 
```
Raspberry Pi OS (32-bit) Lite
Minimal image based on Debian Buster
Version: May 2020
Release date: 2020-05-27
Kernel version: 4.19
Size: 432 MB
```

Rebooting after the scripted install, the Pi does not want to start kmod-static-nodes.service neither systemd-modules-load.service…maybe something unusual happened during the apt-get update/upgrade process…
```
Jun 17 23:31:50 emonpi systemd[1]: Starting Load Kernel Modules...
Jun 17 23:31:50 emonpi systemd-modules-load[1633]: libkmod: ERROR ../libkmod/libkmod-config.c:656 kmod_config_parse: /lib/modules/4.19.118-v7+
Jun 17 23:31:50 emonpi systemd-modules-load[1633]: libkmod: ERROR ../libkmod/libkmod-index.c:787 index_mm_open: magic check fail: 23202f65 ins
Jun 17 23:31:50 emonpi systemd-modules-load[1633]: could not open moddep file '/lib/modules/4.19.118-v7+/modules.dep.bin'
Jun 17 23:31:50 emonpi systemd-modules-load[1633]: Failed to lookup module alias 'i2c-dev': Function not implemented
Jun 17 23:31:50 emonpi systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
Jun 17 23:31:50 emonpi systemd[1]: systemd-modules-load.service: Failed with result 'exit-code'.
Jun 17 23:31:50 emonpi systemd[1]: Failed to start Load Kernel Modules.
```

the system is in recovery mode but the whole emoncms stack is working fine and emoncms can be reached through the network…

kernel version :
```
uname -r
4.19.118-v7+ 
```
it appears there is some disorders in /lib/modules :
```
pi@emonpi:/lib/modules/4.19.118-v7+ $ ls
build                       modules.alias.dpkg-tmp        modules.dep               modules.devname.dpkg-new  modules.softdep.dpkg-tmp
kernel                      modules.builtin               modules.dep.bin           modules.devname.dpkg-tmp  modules.symbols
modules.alias               modules.builtin.bin           modules.dep.bin.dpkg-new  modules.order             modules.symbols.bin
modules.alias.bin           modules.builtin.bin.dpkg-new  modules.dep.bin.dpkg-tmp  modules.order.dpkg-new    modules.symbols.bin.dpkg-new
modules.alias.bin.dpkg-new  modules.builtin.bin.dpkg-tmp  modules.dep.dpkg-new      modules.order.dpkg-tmp    modules.symbols.bin.dpkg-tmp
modules.alias.bin.dpkg-tmp  modules.builtin.dpkg-new      modules.dep.dpkg-tmp      modules.softdep           modules.symbols.dpkg-new
modules.alias.dpkg-new      modules.builtin.dpkg-tmp      modules.devname           modules.softdep.dpkg-new  modules.symbols.dpkg-tmp 
```
so tried to reinstall raspberrypi-kernel-headers but still no progress :
```
sudo apt-get install raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  raspberrypi-kernel-headers
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 24.9 MB of archives.
After this operation, 163 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel-headers armhf 1.20200601-1 [24.9 MB]
Fetched 24.9 MB in 15s (1,708 kB/s)                                                                                                          
Selecting previously unselected package raspberrypi-kernel-headers.
(Reading database ... 48316 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel-headers_1.20200601-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20200601-1) ...
Setting up raspberrypi-kernel-headers (1.20200601-1) ...
```
Anyway, it seems there is no headers package available for my kernel release version
```
sudo apt list linux-headers-*
Listing... Done
linux-headers-4.18.0-3-common-rt/stable 4.18.20-2+rpi1 all
linux-headers-4.18.0-3-common/stable 4.18.20-2+rpi1 all
linux-headers-4.9.0-6-all-armhf/stable 4.9.82-1+deb9u3+rpi2 armhf
linux-headers-4.9.0-6-all/stable 4.9.82-1+deb9u3+rpi2 armhf
linux-headers-4.9.0-6-common-rt/stable 4.9.82-1+deb9u3+rpi2 all
linux-headers-4.9.0-6-common/stable 4.9.82-1+deb9u3+rpi2 all
linux-headers-4.9.0-6-rpi2/stable 4.9.82-1+deb9u3+rpi2 armhf
linux-headers-4.9.0-6-rpi/stable 4.9.82-1+deb9u3+rpi2 armhf
linux-headers-rpi-rpfv/stable 4.9+80+deb9u4+rpi2 armhf
linux-headers-rpi2-rpfv/stable 4.9+80+deb9u4+rpi2 armhf
linux-headers-rpi2/stable 4.9+80+deb9u4+rpi2 armhf
linux-headers-rpi/stable 4.9+80+deb9u4+rpi2 armhf
```
I am not very familiar with all that
@borpin @TrystanLea one of yu maybe know better ?

I  finally could manage to make things work but in a very ugly manner - actually I replaced the whole /lib/modules folder with some content from another pi working fine on the same kernel

Alex

borpin, 2020-06-18:
Ah yes.

As part of the install script, did you copy across the init file?  My guess is that there are changes to the default so we need to add out amendments into the new file.

Basically, we change the partitions created on first boot.

I did think at the time we would need to revisit that!

Can you compare this file with the one in the new image?

https://github.com/openenergymonitor/EmonScripts/blob/ac100a39845cbb4b790afb0c37103c20c8b6683a/install/init_resize.sh#L1

This is all we do to the standard image other than the changes in `config.txt` ref #83

alexandrecuer, 2020-06-20:
> As part of the install script, did you copy across the init file?

Yes I am using the init_resize.sh file coming from the EmonScripts repo, actually the one on the master branch rather than the one on the stable branch, but as I can see, there is no difference

I follow all the process as described here : 
https://github.com/openenergymonitor/EmonScripts/blob/master/install/rpi-install.md

borpin, 2020-06-22:
Ah I mean is, bar the changes to the partition resizing we introduced,  are there now other differences between the Raspbian default file and our resize init file?

alexandrecuer, 2020-06-22:
OK understood....

just catch an original nit_resize.sh and tracked your mods....
here is the result 

https://github.com/dromotherm/sandbox/commit/afa27672980670f045da9ffca57b0e4a301311a8#diff-e373cad2d59c4a4888ff4fa83022324c

dont see much noticeable things except the resizing related ones.....

borpin, 2020-06-22:
Yes the file should be exactly the same but for the partition sizing.

borpin, 2020-06-22:
I made a couple of comments on the diff.

alexandrecuer, 2020-06-25:
Maybe also more related to raspbian than to the Scripts but I had this problem when installing a new SD

```
-------------------------------------------------------------
Setup the Mariadb server (MYSQL)
-------------------------------------------------------------
Reading package lists... Done
Building dependency tree       
Reading state information... Done
mariadb-client is already the newest version (1:10.3.22-0+deb10u1).
mariadb-server is already the newest version (1:10.3.22-0+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
- Secure MYSQL
ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2)
- Create emoncms database
ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2)
- Add user:emoncms and assign to database:emoncms
ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2)
```

alexandrecuer, 2020-06-25:
@borpin : the raspbian resize init file has recently changed 

I have made a better identification of what has been changed in our resize init file

 here is the result

https://github.com/dromotherm/sandbox/commit/dac297a3342921dffe0cde775a563b9fe5dde9b9#diff-f740c31dc7710b1a742fabfe6d6d4645

are all the changes you made present or did I miss something ?

borpin, 2020-06-25:
On MariaDB issue, can you open a new Issue. It was discussed on the community recently but that was the DB in the home directory so should not be an issue for a scripted install.

borpin, 2020-06-25:
> I think we should integrate their changes in our resize init file.

Absolutely. I think you have everything :)

borpin, 2020-07-10:
@alexandrecuer - can you do a PR for this?

@TrystanLea - could this be pushed to master for testing please one done.

alexandrecuer, 2020-07-10:
As far as the resize_init.sh file is concerned, I've done the PR and Trystan has merged it....
https://github.com/openenergymonitor/EmonScripts/commit/8f5baa011b3a52e8f992a0f075756b363c488f84
for the issues related to the kerne & mariadbl, they are not very easily replicable and seem to be sporadic
going to close the issue

borpin, 2020-07-10:
Oh Ok great sorry, missed it.

closed 2020-07-10 by alexandrecuer