openenergymonitor/EmonScripts #111: Python3 not installed

borpin, 2020-09-09:
https://github.com/openenergymonitor/EmonScripts/blob/b1400d10bb8a3ee64ec88fb94e1ff723e1bf38a8/install/main.sh#L64

Only Python not Python3 is installed. Required for service-runner.

Should also be done in the update script.

TrystanLea, 2020-09-22:
Seems that I missed that line, I wonder fi we change to just:

     sudo apt-get install -y git build-essential gettext 

whether that would work correctly, I imagine it would

borpin, 2020-09-22:
No because with just a core install, python3 is never installed now for service runner.

Also, for some distributions Python is installed by default (Ubuntu) and not others (Debian). This may be why we have seen inconsistencies in the past.

We should simply try and install all dependencies, those already satisfied will be ignored.

TrystanLea, 2021-01-21:
I've changed this line now to python3-pip and dev and moved gettext to apache
https://github.com/openenergymonitor/EmonScripts/blob/master/install/main.sh#L64

closed 2021-01-21 by TrystanLea