openenergymonitor/EmonScripts #112: pip/pip3 should not be run as sudo

borpin, 2020-09-09:
https://github.com/openenergymonitor/EmonScripts/blob/cff186a832e44d6692dae5b4217c381f42c94ab7/update/emonhub.sh#L37

pip3 should not be run as sudo (plenty of direction on this in the docs).

TrystanLea, 2020-09-22:
Will that now cause a problem now that we have started with sudo?

borpin, 2020-09-22:
It shouldn't.... (famous last words).

TrystanLea, 2021-01-21:
These have been moved to the emonhub repo but are still installed with sudo
https://github.com/openenergymonitor/emonhub/blob/master/install.sh#L22

Redis client is still installed with sudo, not sure if required...
https://github.com/openenergymonitor/EmonScripts/blob/master/install/redis.sh#L20

TrystanLea, 2021-05-08:
Testing installation on minimalmodbus on a pi yesterday, I noticed that I did need to install with sudo, I think the reason for this is that emonhub is ran under the emonhub username and so cannot access packages installed under pi?

borpin, 2023-01-29:
New image does not use sudo.

closed 2023-01-29 by borpin