openenergymonitor/EmonScripts #113: Do an apt-get install of python3-pip first

borpin, 2020-09-09:
https://github.com/openenergymonitor/EmonScripts/blob/cff186a832e44d6692dae5b4217c381f42c94ab7/update/emonhub.sh#L37

There is no guarantee that pip3 has been installed if updating. Therefore and update/upgrade and python3-pip install should be done.

TrystanLea, 2020-09-22:
There was a note there:

    # Temporary addition of paho-mqtt & requests here
    # remove once issue has cleared:
    # https://community.openenergymonitor.org/t/emonpi-new-sd-image-emonhub-failing/14578
    sudo pip3 install paho-mqtt requests

Can we now remove these lines completely?

Or perhaps we just need to run emonhub/install.sh on each update...?

borpin, 2020-09-22:
No I don't think you can remove them as you don't know what was installed.

There is no harm calling for an install of something. If it is already there the script will simply keep going.

It is also probably worth updating these packages as well I suggest (different issue though).

TrystanLea, 2021-01-21:
This is now included in the emonhub install/update script
https://github.com/openenergymonitor/emonhub/blob/master/install.sh#L21

closed 2021-01-21 by TrystanLea