openenergymonitor/EmonScripts #126: adding wpa_supplicant to the /boot is not enough to bring up a WiFi connection until after init_resize has run

g0lgs, 2020-11-26:
In the Buster Lite Image adding wpa_supplicant.conf to /boot does not allow SSH access until the init_resize and restart has been completed.

You also need to create an empty 'ssh' file in /boot to enable SSH.

Also I believe that the WLAN Country code for the UK is actually GB and not UK (as shown in the sample).

In order to access the Pi to do the changes for init_resize I had to use a Monitor and Keyboard (or you can copy the changed init_resize.sh directly to the SD card if using a Linux machine).

TrystanLea, 2021-01-21:
Thanks @g0lgs 

I've updated the readme to show the wpa_supplicant template as given here:
https://www.raspberrypi.org/documentation/configuration/wireless/headless.md

Step 1 covers SSH I believe.

Adding wpa_supplicant.conf to boot seems to work fine for me for connecting to WiFi?

Please re-open if still an issue

closed 2021-01-21 by TrystanLea

g0lgs, 2021-01-21:
Sorry you seem to have missed part of the problem.

SSH to the Pi when built from the Buster Lite image is NOT possible until init_resize.sh has run - therefore you cannot SSH into the Pi to carry out steps 8 & 9 of https://github.com/openenergymonitor/EmonScripts/blob/master/install/rpi-install.md

g0lgs, 2021-01-21:
I Have just tested using the latest buster lite image from 11 January and your procedure does work - it certainly didn't for me with the previous buster lite image.