openenergymonitor/EmonScripts #142: Use of root user banned

borpin, 2022-03-07:
https://github.com/openenergymonitor/EmonScripts/blob/1a5ed0b90f9bacdb70001c35f1c1c3093a284bc1/update/main.sh#L23-L27

Whilst I can understand the warning, this absolute block should be removed as there may be a good reason why it is being run as root.

glynhudson, 2022-03-07:
I would prefer the block to stay in place. Any user which has the knowledge to run scripts as root will be able to comment out this block. Lots of people prepend 'sudo' onto a script to try to make it run, it causes a lot of issues if it's ran as root. It would be a potential security risk to allow the script to run as root since it can be called via the web.

closed 2022-03-07 by glynhudson

borpin, 2022-03-08:
It is the absolute block that is the issue. Some mechanism to allow it to run would be very useful even by a Y/N if interactive (not service-runner).