openenergymonitor/EmonScripts #148: RaspberryPi OS 32bit Lite install (10th Nov 2022) PROCESS UPDATE

TrystanLea, 2022-11-10:
1\. Default SSH password set using userconf file in /boot

2\.  After running init_resize I then ran firstboot but with the do_resize part commented out
Building on: https://github.com/openenergymonitor/EmonScripts/issues/145 and: https://github.com/openenergymonitor/EmonScripts/issues/147

3\. After completion enabled SPI via raspi-config and one wire temperature sensing support.

4\. Ran steps here: https://github.com/openenergymonitor/EmonScripts/issues/12

**Issues**

wpa_supplicant.conf has wrong permissions, initial WiFi setup UI cannot access it, changed to:

    sudo chmod 644 /etc/wpa_supplicant/wpa_supplicant.conf

TrystanLea, 2022-11-10:
Im getting an error that log2ram cant start due to tmpfs size being too small. It looks like there are some new journal files or logs now in /var/log which are larger than the 40Mb allocated

[image]

TrystanLea, 2022-11-10:
Looks like this must be related to https://forums.raspberrypi.com/viewtopic.php?t=341605

the disk use matches when running:

    journalctl --disk-usage

The procedure described by druck works a treat:

Add the following to /etc/systemd/journald.conf

    Storage=volatile
    RuntimeMaxUse=64M
    ForwardToConsole=no
    ForwardToWall=no

Restart the service

    systemctl restart systemd-journald

Delete the journald log files

    sudo rm -rf /var/log/journal

TrystanLea, 2022-11-11:
Continues from https://github.com/openenergymonitor/EmonScripts/issues/12

after reboot emonpiupdate.log will be present, remove to ensure emonpifirstupdate runs next time... 

1\. Set WiFi country code

    sudo raspi-config

2\. Prepare wifiAP:

    sudo systemctl disable hostapd.service 
    sudo wifiAP start
    sudo reboot

3\. Remove emoncms logs

    sudo rm /var/log/emoncms/emonpiupdate.log
    sudo rm /var/log/emoncms/emonupdate.log
    sudo rm /var/log/emoncms/emoncms.log

4\. Disable SSH

    sudo update-rc.d ssh disable
    sudo invoke-rc.d ssh stop
    sudo dpkg-reconfigure openssh-server
    sudo halt

5\. Check that emonpiupdate.log has been removed before creating image.

6\. Update emonSD-DDMMMYY file on boot partition and in safe-update list on master branch

*/var/log/rotated_logs make cause log2ram to fail??* Remove using SD card reader

7\. Create image:

    sudo dd if=/dev/sdb of=emonSD-10Nov22.img bs=4M

TrystanLea, 2022-11-11:
Other steps:

Enable SPI using raspi-config

Enable DS18B20 support see https://github.com/openenergymonitor/emonhub/tree/master/conf/interfacer_examples/DS18B20

Copy new default.emonhub.conf from emonhub repo

borpin, 2023-01-29:
> Enable DS18B20

This generates errors. This should be added to the docs to enable if required - most will not I suggest.

#156

borpin, 2023-02-05:
TL:DR - just delete the `/var/log/journal` file to fix this issue - no need to edit the `journal.conf`.

> Add the following to /etc/systemd/journald.conf
> 
> ```
> Storage=volatile
> RuntimeMaxUse=64M
> ForwardToConsole=no
> ForwardToWall=no
> ```

Issue with `journald` - still set to `auto` but a `/var/log/journal` folder must be created somewhere along the line on first boot, so the journal is saved to file.

By the time you came to install `log2ram` it was too big!

I looked at this ages ago and perchance I came across this question I asked... https://unix.stackexchange.com/questions/513212/journald-storage-persistent-just-disk-or-ram-disk

You can see what files are in use using this
```
journalctl --unit=systemd-journald --boot 0 --output cat
```

There may be some advantage to saving the `journal` files as they do contain some good info on occasions, however, `ForwardToSyslog=yes` is enabled by default and we preserve this.

Note: adding to the standard file is **always** a bad idea. As we do for other configuration changes, add in a drop-in to the `conf.d` folder and add the command to read it. From the man;
```
When packages need to customize the configuration, they can install configuration snippets in /usr/lib/systemd/*.conf.d/ or /usr/local/lib/systemd/*.conf.d/. The main configuration file is read before any of the configuration directories, and has the lowest precedence;
```

Looking at the commands that have been added, - these 2 are the default so are having no impact.
```
ForwardToConsole=no
ForwardToWall=no
```

Next, the size of this is larger than the size of the `log2ram`  mount (50M)
```
RuntimeMaxUse=64M
```
By default this option will use 10% of the available space, so I think it is mute. Again from the man
```
SystemMaxUse= and RuntimeMaxUse= control how much disk space the journal may use up at most.

The first pair defaults to 10%
```
With this setting we see that:
```
Feb 04 18:48:57 emonpi systemd-journald[345]: Runtime Journal (/run/log/journal/622914b12e1f4c9aaf18d585f863191f) is 8.0M, max 64.0M, 56.0M free.
```

Taking the `RuntimeMaxUse` out results in

```
Feb 05 19:16:19 emonpi systemd-journald[947]: Runtime Journal (/run/log/journal/622914b12e1f4c9aaf18d585f863191f) is 16.0M, max 18.4M, 2.4M free.
```

As you have removed the directory `/var/log/journal/` I should be able to remove the `volatile` statement and let it fall back to `auto` which if there is no folder in `/var/log` will default to volatile.

Restart the daemon and it stays as volatile.

borpin, 2023-02-05:
Fixed via a PR in `log2ram` 

https://github.com/openenergymonitor/log2ram/pull/2

borpin, 2023-02-16:
Outstanding issues;

> wpa_supplicant.conf has wrong permissions
Is this fixed in the script?

> Im getting an error that log2ram cant start due to tmpfs size
Fixed above.

> Other steps:
> 
> Enable SPI using raspi-config
> 
> Enable DS18B20 support

I realised these are specific to the emonPi.

Do we need to do something to enable these via the UI if they need to be connected to make the SDCard more flexible?

I think we should ignore the ARMv6 issue and make a note in the Scripts page.