openenergymonitor/EmonScripts #152: su command in emonhub logrotate file causing issues

borpin, 2022-12-17:
https://github.com/openenergymonitor/EmonScripts/blob/69e87221dacd5fba9b18ed320f9ba9aa5513946e/defaults/etc/logrotate.d/emonhub#L6

The su command is causing issues. 

With it and running `logrotate` without `sudo` this error appears
```
switching euid to 0 and egid to 0
error: error switching euid to 0 and egid to 0: Operation not permitted
```
Removing the command and `logrotate` complains that
```
error: skipping "/var/log/emonhub/emonhub.log" because parent directory has insecure permissions 
(It's world writable or writable by group which is not "root") 
Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
```
Note the parent directory for `emoncms` and `emonhub` are different.
```
drwxr-xr-x  2 pi        root       280 Dec 17 14:52 emoncms
drwxrwxr-x  2 pi        emonhub     60 Jun  9  2020 emonhub
```

However, I cannot see how the `/var/log/emonhub` permission get set to 775!

the use of `su` is a kludge and should be fixed.

alexandrecuer, 2022-12-23:
On the emonhub log folder, user should be the current/service user and group should be root so that logrotate can make its job without `su root root`

The process is very complex for a simple thing :

- The log folder is created by the service file for emonhub.
https://github.com/openenergymonitor/emonhub/blob/afa562bb371a2a426c80ed15a96d808efefc2fed/service/emonhub.service#L15

- If the log folder already exists, emonhub install.sh adjusts the permissions : 
https://github.com/openenergymonitor/emonhub/blob/afa562bb371a2a426c80ed15a96d808efefc2fed/install.sh#L101

I cannot test as I dont have any emonpi with emonhub but at least this line :
https://github.com/openenergymonitor/emonhub/blob/afa562bb371a2a426c80ed15a96d808efefc2fed/install.sh#L109
should be :
```
sudo chown $user:root /var/log/emonhub/emonhub.log
```

But why using the service to create the log ? wont it be enought to just do it in the install.sh file like for other OEM services ?

borpin, 2022-12-23:
Firstly, there are more issues than this on the latest release as discussed here https://community.openenergymonitor.org/t/no-disk-space-although-enough-room/22042/53?u=borpin

The process is because this is part of the Log2ram system, so the ram log files are rotated off the ramlog into persistent disk.

This is also tied up with the emonhub user not being used anymore by emonhub.

There are a multitude of issues and if I get a chance I'll work through it to fix.

closed 2023-02-08 by TrystanLea