openenergymonitor/EmonScripts #153: Assumption of user pi breaks non Pi installs

borpin, 2022-12-17:
https://github.com/openenergymonitor/EmonScripts/blob/69e87221dacd5fba9b18ed320f9ba9aa5513946e/update/main.sh#L60

This breaks an install on any system other than an RPi.

It should use the user in the `config.ini`

https://github.com/openenergymonitor/EmonScripts/blob/69e87221dacd5fba9b18ed320f9ba9aa5513946e/update/main.sh#L23

Breaks an install running as root. It should honour the `user` set in the `config.ini`

bramco, 2022-12-19:
Assume this is a hangover from the days before a new install of Raspbian asks users to set up a user name.

Until the script is fixed, a note on the instructions to say that users must use 'pi' as the user would save a lot of frustration.

borpin, 2022-12-19:
Yes agreed, but the update script should still respect the install config file, no matter what.

borpin, 2023-01-29:
#110 is a duplicate

closed 2023-02-08 by TrystanLea