openenergymonitor/EmonScripts #154: Install errors re Apache2 configuration

borpin, 2022-12-17:
```
Creating config file /etc/php/8.1/apache2/php.ini with new version
Module mpm_event disabled.
Enabling module mpm_prefork.
apache2_switch_mpm Switch to prefork
apache2_reload: Your configuration is broken. Not restarting Apache 2
apache2_reload: AH00112: Warning: DocumentRoot [/var/www/emoncms] does not exist
apache2_reload: (2)No such file or directory: AH02291: Cannot access directory '/var/log/emoncms/' for main error log
apache2_reload: (2)No such file or directory: AH02291: Cannot access directory '/var/log/emoncms/' for error log of vhost defined at /etc/apache2/sites-enabled/emoncms.conf:1
apache2_reload: AH00014: Configuration check failed
apache2_invoke: Enable module php8.1
apache2_reload: Your configuration is broken. Not restarting Apache 2
apache2_reload: AH00112: Warning: DocumentRoot [/var/www/emoncms] does not exist
apache2_reload: (2)No such file or directory: AH02291: Cannot access directory '/var/log/emoncms/' for main error log
apache2_reload: (2)No such file or directory: AH02291: Cannot access directory '/var/log/emoncms/' for error log of vhost defined at /etc/apache2/sites-enabled/emoncms.conf:1
apache2_reload: AH00014: Configuration check failed
```

I suspect this is caused as Apache is installed an configured before emoncms is installed.

It is not fatal, but a good install would have no errors.

bramco, 2022-12-19:
It doesn't seem to affect the system running though, so as per needing to set up the user as 'pi', a note could be put in the instructions until this is fixed.

borpin, 2022-12-19:
Yes that was my point. the `vhost` config needs to be done later, or the folders created earlier.

closed 2023-02-08 by TrystanLea