openenergymonitor/EmonScripts #157: content of apache2.error.log

alexandrecuer, 2023-01-10:
Does somebody has got this kind of things at boot in the apache2.error.log ?
```
cat: /sys/devices/virtual/dmi/id/board_vendor: No such file or directory
cat: /sys/devices/virtual/dmi/id/product_name: No such file or directory
cat: /sys/devices/virtual/dmi/id/board_name: No such file or directory
cat: /sys/devices/virtual/dmi/id/bios_version: No such file or directory
cat: /sys/devices/virtual/dmi/id/bios_date: No such file or directory
sh: 1: iostat: not found
sh: 1: iostat: not found
sh: 1: iostat: not found
sh: 1: iostat: not found
sh: 1: iostat: not found
sh: 1: iostat: not found
sh: 1: iostat: not found
sh: 1: iostat: not found
sh: 1: iostat: not found
sh: 1: iostat: not found
sh: 1: iostat: not found
sh: 1: /opt/vc/bin/vcgencmd: not found
[Mon Jan 09 18:06:39.275617 2023] [php7:notice] [pid 3633] [client 192.168.3.4:37148] PHP Notice:  Undefined index: hw in /var/www/emoncms/Modules/admin/Views/admin_main_view.php on line 224, referer: http://192.168.4.5/graph/27
```

borpin, 2023-01-29:
@alexandrecuer - what base OS / Build , PHP version etc.

What were you doing to generate the errors?

alexandrecuer, 2023-02-01:
raspberrypiOS 64 bits.
all details here https://dromotherm.github.io/documentation/pages/release_notes.txt
I have a few 32 bits OS running and it is true they do not have these errors

borpin, 2023-02-01:
Yes it is a 64bit error. We only support a build on 32bit currently.

Happy for you to find the solution :)

closed 2024-10-14 by alexandrecuer