openenergymonitor/EmonScripts #159: Once `/var/log` fills up the update can no longer run

borpin, 2023-01-29:
#85 solves the issue of logrotate not rotating it's own logs. However it does not solve the issue of once `/var/log` fills up the update can no longer run.

@TrystanLea what do you think of adding a check right at the beginning of the update script to check if if /var/log is full and if so truncate the largest log? As suggested by @borpin [on this thread](https://community.openenergymonitor.org/t/disk-space-for-logs/14392/18?u=glyn.hudson). Or maybe use a different location for emonpiupdate.log?

_Originally posted by @glynhudson in https://github.com/openenergymonitor/EmonScripts/issues/81#issuecomment-642104437_

TrystanLea, 2023-01-31:
Ok so will still need a script of some sort to check and clear logs if full ? Should the logrotate config not fix this by ensuring that the sum of all logs cannot exceed the log partition size?

borpin, 2023-02-04:
Ha, yes there could be - had an emonhub log fill up :(

CDuffers, 2023-03-26:
Morning,

Checking and clearing space before we run is always a good thing to do.

I suggest that we need to enhance the error checking through the scripts to pick up this and any other quirks and foibles along the way.

I'm working on an enhancement to `emoncms-export.sh` within the backup module, my current approach is to run a trap on error which shows the RC and LINENO before allowing it to continue, then also a trap on EXIT to summarise if it completed successfully or with error. The idea being to highlight a problem rather than stop and perform some more invasive controls, which I thought was too strong a change at this point.

This might be helpful for the install and update scripts as well as it MIGHT have highlighted the root cause from this issue https://github.com/openenergymonitor/EmonScripts/issues/179

This is likely to requite rethinking of some code to prevent non-zero return codes for commands like grep when it can't find something. e.g. `$(ls /boot | grep emonSD)` when run on my Ubuntu VM