openenergymonitor/EmonScripts #169: Apt upgrade not done during update

borpin, 2023-02-04:
https://github.com/openenergymonitor/EmonScripts/blob/850099f87a699a3b5ac6516ef5d8e3a312559ec0/update/main.sh#L66

An `apt update` is done, but not an 'apt upgrade` during the update process.

On my old system, I had 600 odd packages to update.

A fresh Nov22 image had 60 when I installed it today.

These should be maintained, I believe.

TrystanLea, 2023-02-08:
The concern here is the time taken and uncertainty on the implications of upgrading packages this way. Perhaps this needs to be a separate OS package upgrade option via the UI

borpin, 2023-02-09:
That sounds like a plan. Perhaps with a note on how many packages are outstanding?

Can we feed that update to a log file/window?

I do think it is needed.

borpin, 2023-02-10:
I think a check could be done if it is a Nov22 Image, then do the update.

For each new image, check and update if latest image.