openenergymonitor/EmonScripts #171: a possible solution for removing the use of sudo with pip

alexandrecuer, 2023-02-07:
@borpin often writes not to use pip with sudo and he is very right.

For me, the use of sudo with pip was to install package globally and I thought it was ok....

This is a bad practise because we are using a binary distrubtion which comes with some base packages already installed, sometimes some of them disappear, sometimes new ones appear....
Anyway there is a simple workaround but I was not aware it was simple.
```
sudo apt-get install python3-venv
python3 -m venv /opt/oem
```
the 2 previous lines permit to create a venv named oem located on /opt
Now you have to install the package in that venv
for that you activate the venv then install through pip, and only pip, so you dont mix python3 packages coming from apt and others coming from pip, which is a dirty headache to maintain from my point of view
```
cd /opt/oem/bin
source activate
python3 -m pip install pyserial
python3 -m pip install smbus
python3 -m pip install gpiozero
python3 -m pip install paho-mqtt
python3 -m pip install redis
etc
etc
```
to exit the venv :
```
deactivate
```
now the only thing you have to do is to change the shebang in the python exe file.
for example `#!/usr/bin/env python3` has to be replaced by `#!/opt/oem/bin/python3`

borpin, 2023-02-10:
Yes we have considered `venv` but it is not something either @TrystanLea or myself are familiar with. I understand the basic concept, but nothing more than that. I've done some work to include the ESPHome console onto the emonSD and this involved a venv. Making that run correctly from a service file was a little challenging and I'm still unclear how to correctly update a venv from a script.

If you want to work on a working example, I'm sure we would look at it again.

It would be a much cleaner solution :)

alexandrecuer, 2023-02-16:
@borpin @TrystanLea : a working example is here

https://github.com/openenergymonitor/emonhub/pull/184

tested on an emonpi with an emontx v3 with emontxcm firmware (not updated to the latest)