openenergymonitor/EmonScripts #179: Ubunto 20.4 Error Emoncms install script : line 52 command not found

buchse12, 2023-02-26:
Hello,
I've been trying to fix the problem for many hours without success.
Proxmox LXC container created with Ubunto 20.4 and the Emoncms instructions all executed and accordingly adjusted the config.ini.
The installation does not go through and gives the following error:

`Cloning into 'usefulscripts'...
remote: Enumerating objects: 1034, done.
remote: Counting objects: 100% (136/136), done.
remote: Compressing objects: 100% (77/77), done.
remote: Total 1034 (delta 60), reused 124 (delta 57), pack-reused 898
Receiving objects: 100% (1034/1034), 284.07 KiB | 4.30 MiB/s, done.
Resolving deltas: 100% (522/522), done.
Update Emoncms database
/opt/openenergymonitor/EmonScripts/install/emoncms_modules.sh: line 52: php: com                                                                                                    mand not found
root@emoncms:/opt/openenergymonitor/EmonScripts/install#

can someone help me? I am in despair here.

`

buchse12, 2023-02-28:
Hello,
Can no one give a tip why the script aborts with this error?
``/opt/openenergymonitor/EmonScripts/install/emoncms_modules.sh: line 52: php: com mand not found``

CDuffers, 2023-03-26:
Hi @buchse12 ,

Please try sending problems like this to the community linked below as people look there regularly:
https://community.openenergymonitor.org/

Did you make any changes to the `config.ini`? In particular do you still have a line `install_php=true`? And without a `#` at the beginning?

If it is set, then earlier in the install output you should see a section similar to the following:

```-------------------------------------------------------------
Install PHP
-------------------------------------------------------------
not ARMv6 architecture OK to use PHP8.1
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libapache2-mod-php8.1 php-common php8.1-cli php8.1-common php8.1-opcache
  php8.1-readline
Suggested packages:
  php-pear
The following NEW packages will be installed
  libapache2-mod-php8.1 php-common php8.1 php8.1-cli php8.1-common
  php8.1-opcache php8.1-readline
0 to upgrade, 7 to newly install, 0 to remove and 1 not to upgrade.
Need to get 5,149 kB of archives.
```

My presumption is that there has been a problem installing php in this step.

Have you run the install as root? If so, then I'd suggest creating a user to run this under that instead.

Have you seen the docs here?
https://docs.openenergymonitor.org/emonsd/install.html

sandyman44, 2023-04-18:
I get the exact same issue. installing on ubuntu 20.04 server on azure. 

`/opt/openenergymonitor/EmonScripts/install/emoncms_modules.sh: line 52: php: command not found`
yes I am following the docs at https://docs.openenergymonitor.org/emonsd/install.html

I speculatively did `sudo apt install php` which went off and did its thing, but I now get this

```
-------------------------------------------------------------
Install Emoncms Core Modules
-------------------------------------------------------------
- Module dashboard already exists
- Module device already exists
- Module app already exists
- Module graph already exists
- Module postprocess already exists
- Module demandshaper already exists
- Module backup already exists
- Module sync already exists
- Module usefulscripts already exists
Update Emoncms database
PHP Warning:  PHP Startup: Unable to load dynamic library 'mosquitto.so' (tried: /usr/lib/php/20190902/mosquitto.so (/usr/lib/php/20190902/mosquitto.so: cannot open shared object file: No such file or directory), /usr/lib/php/20190902/mosquitto.so.so (/usr/lib/php/20190902/mosquitto.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
PHP Warning:  PHP Startup: Unable to load dynamic library 'redis.so' (tried: /usr/lib/php/20190902/redis.so (/usr/lib/php/20190902/redis.so: cannot open shared object file: No such file or directory), /usr/lib/php/20190902/redis.so.so (/usr/lib/php/20190902/redis.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
PHP Fatal error:  Uncaught Error: Class 'mysqli' not found in /opt/openenergymonitor/EmonScripts/common/emoncmsdbupdate.php:11
Stack trace:
#0 {main}
  thrown in /opt/openenergymonitor/EmonScripts/common/emoncmsdbupdate.php on line 11

Fatal error: Uncaught Error: Class 'mysqli' not found in /opt/openenergymonitor/EmonScripts/common/emoncmsdbupdate.php:11
Stack trace:
#0 {main}
  thrown in /opt/openenergymonitor/EmonScripts/common/emoncmsdbupdate.php on line 11
```

CDuffers, 2023-04-23:
Hi @sandyman44 

Out of curiosity, is there a reason why you're not using 22.04?

This looks to be a symptom rather than the cause, you'll need to check through the earlier stages of the output to confirm what's happening.

On the presumption that your `config.ini` has `install_php=true` and `install_mosquitto=true`, `install_mosquitto_server=true` and `install_mosquitto_client=true`, then please check for for errors in the install PHP section I highlight in the previous message and also for the mosquitto install, an example from my install:

```
-------------------------------------------------------------
Mosquitto Server installation and configuration
-------------------------------------------------------------
Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  linux-headers-5.19.0-21 linux-headers-5.19.0-21-generic
  linux-image-5.19.0-21-generic linux-modules-5.19.0-21-generic
  linux-modules-extra-5.19.0-21-generic
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libcjson1 libdlt2 libmosquitto1
The following NEW packages will be installed
  libcjson1 libdlt2 libmosquitto1 mosquitto
```

sandyman44, 2023-04-23:
why not 22.04 - I just went to azure marketplace and selected the latest version of ubuntu that popped up when I went to create a vm - which was 20.04. I've stopped caring about different versions of linux any more its too timeconsuming/tedious for someone who is a "user" rather than a "dev". Having gone back into azure marketplace, it seems 22 is there but doesn't pop up by default you have to search for it.

eventually I figured out for myself that the emoncms was expecting PHP8.1, which is not the version that was getting installed when I tried to install php from that earlier version of ubuntu. I got the same behaviour with debian 11 (Again, the default in azure).   I suspect this is what @buchse12 was hitting as well. either no php, or wrong version of php, either way, not working.

anyway I followed instructions that I found elsewhere on the internet, to connect an extra repository and install PHP8.1, after doing that, the emoncms install worked fine from debian 11. I haven't been back to try it with ubuntu 22 as I now have a working system.

I'd suggest that the emoncms scripts and install documention could do with pointers or notes about  specific versions of ubuntu / debian and specific versions of php for the unwary such as me!

borpin, 2023-05-26:
> eventually I figured out for myself that the emoncms was expecting PHP8.1

I think this has now been solved by changes to the script.