openenergymonitor/EmonScripts #200: Error: `emonSD base image old or undefined with image emonSD-06Jan25`

mnbf9rca, 2025-03-14:
I raised this as openenergymonitor/emonpi#150 but it doesnt seem to have been noticed.

When attempting to check for updates, i get this error:
```
Starting update via service-runner-update.sh (v3.0) >
Filesystem      Size  Used Avail Use% Mounted on
udev            1.6G     0  1.6G   0% /dev
tmpfs           380M   39M  341M  11% /run
/dev/mmcblk0p2  4.4G  3.4G  856M  80% /
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           5.0M   16K  5.0M   1% /run/lock
tmpfs            30M   28K   30M   1% /tmp
tmpfs           1.0M   12K 1012K   2% /var/lib/php/sessions
tmpfs           128M     0  128M   0% /var/tmp
/dev/mmcblk0p1  510M   80M  431M  16% /boot/firmware
/dev/mmcblk0p3  9.8G  9.9M  9.3G   1% /var/opt/emoncms
log2ram          50M  3.9M   47M   8% /var/log
- emonSD version: emonSD-06Jan25
- supported images: emonSD-01Feb24 emonSD-20Nov23 emonSD-10Nov22 emonSD-21Jul21 emonSD-08May21 emonSD-24Jul20 emonSD-02Oct19 emonSD-17Oct19
ERROR: emonSD base image old or undefined...update will not continue
See latest verson: https://github.com/openenergymonitor/emonpi/wiki/emonSD-pre-built-SD-card-Download-&-Change-Log
Stopping update
```

borpin, 2025-03-18:
@TrystanLea The safe-update list is not correct. Missing 2 images (at least)

How has this not been noticed before!

borpin, 2025-04-14:
The 'bug' is that when using the scripts, it writes a file for Today's date.

@TrystanLea the solution may be
When using the Script
1. To parse the safe build list and extract the last date and use that.
2. Add an additional element to the end to indicate whether it was an EmonSD build or a scripts build.

When updating
1. The parsing just checks the date part.